esp-rs/espflash #868: esp32c6 flash reboot loop "Failed to fetch app description header" / "Factory app partition is not bootable"

thecubic, 2025-05-26:
After flashing a known good Rust program (tested successful with `probe-rs` on two Xiao C6s), a `run --release`  with `espflash` causes the device (a C6 DevKit-C) to fail-to-boot and loop infinitely

I'm using `espflash 4.0.0-dev` as of `628947d`

```
$ cargo run --release
    Finished `release` profile [optimized + debuginfo] target(s) in 0.12s
     Running `espflash flash --monitor target/riscv32imac-unknown-none-elf/release/blinky`
[2025-05-26T18:22:59Z INFO ] Serial port: '/dev/ttyUSB0'
[2025-05-26T18:22:59Z INFO ] Connecting...
[2025-05-26T18:22:59Z INFO ] Using flash stub
Chip type:         esp32c6 (revision v0.1)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi 6, BT 5
MAC address:       [REDACTED]
App/part. size:    563,136/8,323,072 bytes, 6.77%
[00:00:01] [========================================]      14/14      0x0                                                                                                                                                                    
[00:00:00] [========================================]       1/1       0x8000                                                                                                                                                                 
[00:00:32] [========================================]     331/331     0x10000                                                                                                                                                                [2025-05-26T18:23:36Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe90
load:0x4086e610,len:0x3054
entry 0x4086c11a
I (23) boot: ESP-IDF v5.4.1-426-g3ad36321ea 2nd stage bootloader
I (24) boot: compile time Apr 29 2025 15:14:17
I (24) boot: chip revision: v0.1
I (25) boot: efuse block revision: v0.2
I (28) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 8MB
I (40) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (47) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (60) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (66) boot:  2 factory          factory app      00 00 00010000 007f0000
I (73) boot: End of partition table
E (76) esp_image: Failed to fetch app description header!
E (81) boot: Factory app partition is not bootable
E (86) boot: No bootable app partitions in the partition table
ESP-ROM:esp32c6-20220919
```

skibon02, 2025-07-04:
I have the same issue.
However, `espflash v3.3.0` works fine

When I add `esp-bootloader-esp-idf` as suggested by @jessebraham I got linker error:
```
  = note: [ldproxy] Running ldproxy
          Error: Linker /home/user/projects/esp32-freq-measure/.embuild/espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-gcc failed: exit status: 1
          STDERR OUTPUT:
          /home/user/projects/esp32-freq-measure/.embuild/espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/newlib/libnewlib.a(assert.c.obj): in function `__assert_func':
          /home/user/projects/esp32-freq-measure/.embuild/espressif/esp-idf/v5.3.2/components/newlib/assert.c:33: multiple definition of `__assert_func'; /home/user/projects/esp32-freq-measure/target/riscv32imac-esp-espidf/release/deps/libesp_rom_sys-33b0d4242521809f.rlib(esp_rom_sys-33b0d4242521809f.esp_rom_sys.39b0d4fe2ffa2aa2-cgu.0.rcgu.o):esp_rom_sys.39b0d4fe2ffa2aa2-cgu.0:(.text.unlikely.__assert_func+0x0): first defined here
          /home/user/projects/esp32-freq-measure/.embuild/espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: The gap between .flash.appdesc and .flash.rodata must not exist to produce the final bin image.
          collect2: error: ld returned 1 exit status
```

project is generated by `esp-idf-template`, and depends on 
```
ESP_IDF_VERSION = "v5.3.2"
```

skibon02, 2025-07-04:
Can be reproduced with default `esp-idf-template` generated project when built without optimizations, looks like behavior depends on the binary size

SergioGasquez, 2025-07-07:
`esp-bootloader-esp-idf` is used for no_std apps (`esp-hal` based), since you are building an std app, you should use the std version of the macro, I think its `esp_idf_sys::esp_app_desc`.

Can you try running `esptool flash_id` on your devkit? The fact that espflash@3.3 and 4.0 give a different spi flash size is weird

skibon02, 2025-07-07:
Yeah, probably I should've created a new issue for this

1. `flash_id output`
```
$ esptool.py --chip esp32c6 --port /dev/ttyACM0 flash_id
esptool.py v4.8.1
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-C6 (QFN40) (revision v0.1)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: f0:f5:bd:ff:fe:10:49:48
BASE MAC: f0:f5:bd:10:49:48
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Manufacturer: 85
Device: 2018
Detected flash size: 16MB
Hard resetting via RTS pin...
```

2. Running `espflash@4.0.0` for the fresh esp-idf-template project
```
$ espflash flash --monitor target/riscv32imac-esp-espidf/release/remote-tester 
[2025-07-07T08:27:27Z INFO ] Serial port: '/dev/ttyACM0'
[2025-07-07T08:27:27Z INFO ] Connecting...
[2025-07-07T08:27:27Z INFO ] Using flash stub
Chip type:         esp32c6 (revision v0.1)
Crystal frequency: 40 MHz
Flash size:        16MB
Features:          WiFi 6, BT 5
MAC address:       f0:f5:bd:10:49:48
App/part. size:    406,576/16,384,000 bytes, 2.48%
[00:00:01] [========================================]      14/14      0x0      Verifying... OK!            
[00:00:00] [========================================]       1/1       0x8000   Verifying... OK!            
[00:00:24] [========================================]     233/233     0x10000  Verifying... OK!                                                                                                                                     
[2025-07-07T08:27:54Z INFO ] Flashing has completed!                                                                                                                                                                                
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe90
load:0x4086e610,len:0x3054
entry 0x4086c11a
I (23) boot: ESP-IDF v5.4.1-426-g3ad36321ea 2nd stage bootloader
I (24) boot: compile time Apr 29 2025 15:14:17
I (24) boot: chip revision: v0.1
I (25) boot: efuse block revision: v0.3
I (28) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 16MB
I (40) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (47) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (60) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (66) boot:  2 factory          factory app      00 00 00010000 00fa0000
I (73) boot: End of partition table
E (76) esp_image: Segment 0 load address 0x42048020, doesn't match data 0x00010020
E (83) boot: Factory app partition is not bootable
E (88) boot: No bootable app partitions in the partition table
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x3 (LP_SW_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001974a
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe90
load:0x4086e610,len:0x3054
entry 0x4086c11a
I (24) boot: ESP-IDF v5.4.1-426-g3ad36321ea 2nd stage bootloader
I (24) boot: compile time Apr 29 2025 15:14:17
I (25) boot: chip revision: v0.1
I (25) boot: efuse block revision: v0.3
I (29) boot.esp32c6: SPI Speed      : 80MHz
I (33) boot.esp32c6: SPI Mode       : DIO
I (37) boot.esp32c6: SPI Flash Size : 16MB
I (40) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (54) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (60) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (67) boot:  2 factory          factory app      00 00 00010000 00fa0000
I (74) boot: End of partition table
E (77) esp_image: Segment 0 load address 0x42048020, doesn't match data 0x00010020
E (84) boot: Factory app partition is not bootable
E (88) boot: No bootable app partitions in the partition table
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x3 (LP_SW_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001974a
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe90
load:0x4086e610,len:0x3054
entry 0x4086c11a
I (24) boot: ESP-IDF v5.4.1-426-g3ad36321ea 2nd stage bootloader
```
... endless reboots ...

MabezDev, 2025-07-07:
Can you try setting the flash speed manually back to 40mhz? I think `-f 40mhz` should do it.

SergioGasquez, 2025-07-07:
Just tried generating a fresh esp-idf-template project(with default values) and flashing an ESP32-C6-DevKitC-1 with no issues: 
<details>
  <summary>Full log</summary>

```
> cargo r -r
    Finished `release` profile [optimized] target(s) in 0.08s
     Running `espflash flash --monitor target/riscv32imac-esp-espidf/release/app-des`
[2025-07-07T10:22:28Z INFO ] Serial port: '/dev/cu.usbserial-1120'
[2025-07-07T10:22:28Z INFO ] Connecting...
[2025-07-07T10:22:28Z INFO ] Using flash stub
[2025-07-07T10:22:29Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32c6 (revision v0.0)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi 6, BT 5
MAC address:       60:55:f9:f7:1b:6c
App/part. size:    382,384/8,323,072 bytes, 4.59%
[00:00:00] [========================================]      14/14      0x0      Verifying... OK!                            [00:00:00] [========================================]       1/1       0x8000   Verifying... OK!                            [00:00:06] [========================================]     199/199     0x10000  Verifying... OK!                            [2025-07-07T10:22:36Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe90
load:0x4086e610,len:0x3054
entry 0x4086c11a
I (23) boot: ESP-IDF v5.4.1-426-g3ad36321ea 2nd stage bootloader
I (24) boot: compile time Apr 29 2025 15:14:17
I (24) boot: chip revision: v0.0
I (25) boot: efuse block revision: v0.0
I (28) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 8MB
I (40) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (47) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (60) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (66) boot:  2 factory          factory app      00 00 00010000 007f0000
I (73) boot: End of partition table
I (76) esp_image: segment 0: paddr=00010020 vaddr=42040020 size=11db0h ( 73136) map
I (97) esp_image: segment 1: paddr=00021dd8 vaddr=40800000 size=0922ch ( 37420) load
I (106) esp_image: segment 2: paddr=0002b00c vaddr=40809230 size=01bdch (  7132) load
I (108) esp_image: segment 3: paddr=0002cbf0 vaddr=00000000 size=03428h ( 13352) 
I (112) esp_image: segment 4: paddr=00030020 vaddr=42000020 size=3d568h (251240) map
I (165) boot: Loaded app from partition at offset 0x10000
I (166) boot: Disabling RNG early entropy source...
I (175) cpu_start: Unicore app
W (185) clk: esp_perip_clk_init() has not been implemented yet
I (192) cpu_start: Pro cpu start user code
I (192) cpu_start: cpu freq: 160000000 Hz
I (192) cpu_start: Application information:
I (195) cpu_start: Project name:     libespidf
I (200) cpu_start: App version:      1
I (204) cpu_start: Compile time:     Jul  7 2025 12:13:58
I (210) cpu_start: ELF file SHA256:  000000000...
I (216) cpu_start: ESP-IDF:          v5.2.3
I (221) cpu_start: Min chip rev:     v0.0
I (225) cpu_start: Max chip rev:     v0.99 
I (230) cpu_start: Chip rev:         v0.0
I (235) heap_init: Initializing. RAM available for dynamic allocation:
I (242) heap_init: At 4080BE20 len 000707F0 (449 KiB): RAM
I (248) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (254) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
[... 23 lines cut ...]
W (441) eFuse: calibration efuse version does not match, set default version to 0
W (449) eFuse: calibration efuse version does not match, set default version to 0
W (457) eFuse: calibration efuse version does not match, set default version to 0
W (465) eFuse: calibration efuse version does not match, set default version to 0
W (481) eFuse: calibration efuse version does not match, set default version to 0
W (482) eFuse: calibration efuse version does not match, set default version to 0
W (490) eFuse: calibration efuse version does not match, set default version to 0
W (498) eFuse: calibration efuse version does not match, set default version to 0
W (506) eFuse: calibration efuse version does not match, set default version to 0
W (514) eFuse: calibration efuse version does not match, set default version to 0
W (522) eFuse: calibration efuse version does not match, set default version to 0
W (531) pcnt(legacy): legacy driver is deprecated, please migrate to `driver/pulse_cnt.h`
W (539) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
W (550) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (559) sleep: Configure to isolate all GPIO pins in sleep state
I (565) sleep: Enable automatic switching of GPIO sleep configuration
I (572) coexist: coex firmware version: 27d8387
I (578) coexist: coexist rom version 5b8dcfa
I (583) main_task: Started on CPU0
I (583) main_task: Calling app_main()
I (583) app_des: Hello, world!
I (593) main_task: Returned from app_main()
```

</details>

skibon02, 2025-07-07:
It just skipped reflashing step:

```
$ espflash flash -f 40mhz --monitor target/riscv32imac-esp-espidf/release/remote-tester
[2025-07-07T10:26:36Z INFO ] Serial port: '/dev/ttyACM0'
[2025-07-07T10:26:36Z INFO ] Connecting...
[2025-07-07T10:26:36Z INFO ] Using flash stub
Chip type:         esp32c6 (revision v0.1)
Crystal frequency: 40 MHz
Flash size:        16MB
Features:          WiFi 6, BT 5
MAC address:       f0:f5:bd:10:49:48
App/part. size:    406,576/16,384,000 bytes, 2.48%
[00:00:00] [========================================]      14/14      0x0      Skipped! (checksum matches)                                                                                                                          
[00:00:00] [========================================]       1/1       0x8000   Skipped! (checksum matches)                                                                                                                          
[00:00:00] [========================================]     233/233     0x10000  Skipped! (checksum matches)
```
... endless reboots

Trying different frequency
```
$ espflash flash -f 80mhz --monitor target/riscv32imac-esp-espidf/release/remote-tester
[2025-07-07T10:27:27Z INFO ] Serial port: '/dev/ttyACM0'
[2025-07-07T10:27:27Z INFO ] Connecting...
[2025-07-07T10:27:27Z INFO ] Using flash stub
Chip type:         esp32c6 (revision v0.1)
Crystal frequency: 40 MHz
Flash size:        16MB
Features:          WiFi 6, BT 5
MAC address:       f0:f5:bd:10:49:48
App/part. size:    406,576/16,384,000 bytes, 2.48%
[00:00:01] [========================================]      14/14      0x0      Verifying... OK!                                                                                                                                     
[00:00:00] [========================================]       1/1       0x8000   Skipped! (checksum matches)                                                                                                                          
[00:00:24] [========================================]     233/233     0x10000  Verifying... OK!                                                                                                                                     
[2025-07-07T10:27:53Z INFO ] Flashing has completed!                                                                                                                                                                                
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe90
load:0x4086e610,len:0x3054
entry 0x4086c11a
I (23) boot: ESP-IDF v5.4.1-426-g3ad36321ea 2nd stage bootloader
I (24) boot: compile time Apr 29 2025 15:14:17
I (24) boot: chip revision: v0.1
I (25) boot: efuse block revision: v0.3
I (28) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 16MB
I (40) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (47) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (60) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (66) boot:  2 factory          factory app      00 00 00010000 00fa0000
I (73) boot: End of partition table
E (76) esp_image: Segment 0 load address 0x42048020, doesn't match data 0x00010020
E (83) boot: Factory app partition is not bootable
E (88) boot: No bootable app partitions in the partition table
```

SergioGasquez, 2025-07-09:
@skibon02 have you tried using `espflash@4.0.1`? I just tried your reproducer and it worked for me! If you are still having issues, could you try using adding `--bootloader target/riscv32imac-esp-espidf/debug/bootloader.bin` to the `espflash` command in `.cargo/config.toml`

skibon02, 2025-07-09:
`$ espflash flash --bootloader target/riscv32imac-esp-espidf/release/bootloader.bin --chip esp32c6 --monitor target/riscv32imac-esp-espidf/release/remote-tester`

It works indeed! When i added bootloader flag it executed reprogramming for a  bootloader sector and it no longer reboots.

However, without `--bootloader` it still doesn't work for both 4.0.0 and 4.0.1

SergioGasquez, 2025-07-10:
Nice! I would recommend you use the bootloader and partition table esp-idf generates (that is, the ones under `target/riscv32imac-esp-espidf/<build_mode>`)

> However, without --bootloader it still doesn't work for both 4.0.0 and 4.0.1 (in release builds on that repository)

I see, for some reason, I could flash your reproducer in debug mode with both 4.0.0 and 4.0.1.

skibon02, 2025-07-10:
Interesting
Maybe it is possible to select bootloader automatically if the project is esp-idf project? Before 4.0.0 it was just working, and now it can confuse a large portion of users

skibon02, 2025-07-11:
I've updated [README.md](https://github.com/skibon02/esp32c6-bootloader-issue/blob/main/README.md) of my reproducer project to include everything known so far

hrl10, 2025-09-28:
my m5stack nanoc6 seems got familiar problem (different flavor "checksum failure")
didn't work with specific bootloader and worked normally with arduino ide

closed 2025-09-28 by thecubic