esp-rs/espflash #873: Avoid leaking inner error types

jessebraham, 2025-06-02:
See the *"Common mistake: inner error type leakage"* section of this blog post for more information:  
https://d34dl0ck.me/rust-bites-designing-error-types-in-rust-libraries/index.html

Given the relatively small amount of effort required, I think it's worth doing this. We can create a type alias for the `Box< .. >` mess and just use that everywhere that we are currently leaking error types.

MabezDev, 2025-06-24:
Oops, I think some of those were added after the initial PR by me 😅. Do you know if there is a lint or check we can do to prevent PRs being merged like this?

bugadani, 2025-06-25:
[somewhat related rustc issue](https://github.com/rust-lang/rust/issues/120149)