esp-rs/espflash #926: Binary flashed with v4 doesn't return from `esp_hal::init` when PSRAM is enabled

robertbastian, 2025-07-07:
* A binary without the `esp_hal/psram` feature flashed with v4 works
* A binary __with__ the `esp_hal/psram` feature flashed with v4 doesn't return from `esp_hal::init`
* Both binaries work with espflash v3

I'm using `esp_hal` 1.0.0-beta.1.

I can try to get a minimal repro but without my exact board it might not be possible to reproduce. Any pointers? This may be related to the new bootloader, is there an issue tracker for that?

robertbastian, 2025-07-07:
Probably but I don't think I get a panic

kendra-pdx, 2025-07-07:
> Is this the same issue as [esp-rs/esp-hal#3765](https://github.com/esp-rs/esp-hal/issues/3765) ?

Looks pretty similar to me. My panic is coming from here: https://github.com/esp-rs/esp-hal/blob/692a768b3aaa809fab861804c4b187f663f8df75/esp-hal/src/soc/esp32s3/psram.rs#L231

robertbastian, 2025-07-07:
I'll close this because the other issue has more detail.

closed 2025-07-07 by robertbastian