esp-rs/espflash #928: espflash monitor Error:   × The bootloader returned an error

daychongyang, 2025-07-14:
I am a beginner at ESP, but I just failed to execute the first command on my **ESP32-C3** device, and what frustrates me is that I can't find the specific reason.

daychongyang, 2025-07-22:
I connected the new ESP32 board to the computer via USB, can I directly obtain the corresponding device information using the following command: `espflash board-info`?

JurajSadel, 2025-07-22:
> I connected the new ESP32 board to the computer via USB, can I directly obtain the corresponding device information using the following command: `espflash board-info`?

Yes, you can.
> 
> I am quite confused now, unable to confirm which link has a problem (is it that the ESP32 board is not supported or there is an issue with the debugging environment), and I always cannot get normal feedback on the terminal.

ESP32 is supported.

```rust
espflash board-info
[2025-07-22T08:04:20Z INFO ] Serial port: '/dev/cu.usbserial-1110'
[2025-07-22T08:04:20Z INFO ] Connecting...
[2025-07-22T08:04:20Z INFO ] Using flash stub
[2025-07-22T08:04:21Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32 (revision v3.0)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
MAC address:       34:86:5d:dc:2c:40
Security features: None
```

daychongyang, 2025-07-22:
Do I need to perform any special initialization on the newly purchased board? Or can I execute related commands and communicate with it after the USB connection is established?

JurajSadel, 2025-07-23:
Can you provide additional information about your project? How did you create it, what dependencies are you using etc?