esp-rs/espflash #938: Main thread panic on flash (?)

Doloro1978, 2025-08-10:
Hihi, im trying to flash my esp32 dev board but i keep running into an error

```
Running `espflash flash --monitor target/xtensa-esp32-espidf/debug/ir`
[2025-08-10T14:41:39Z INFO ] Serial port: '/dev/ttyUSB0'
[2025-08-10T14:41:39Z INFO ] Connecting...
[2025-08-10T14:41:39Z INFO ] Using flash stub
Error:   × Main thread panicked.
  ├─▶ at /home/doloro/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/serialport-4.7.2/src/posix/tty.rs:499:9
  ╰─▶ assertion failed: termios2.c_ospeed == termios2.c_ispeed
  help: set the `RUST_BACKTRACE=1` environment variable to display a backtrace.
```

I'm on espflash 4.0.1 with Linux 6.15.9-arch1-1

LLeny, 2025-08-13:
Facing the same issue with espflash 4.0.1 and Linux 6.15.9-arch1-1. It's a recent issue.
`[2025-08-13T04:05:27Z DEBUG] Flash(
        FlashArgs {
            connect_args: ConnectArgs {
                after: HardReset,
                baud: Some(
                    9600,
                ),
                before: DefaultReset,
                chip: None,
                confirm_port: false,
                list_all_ports: false,
                no_stub: false,
                port: None,
                non_interactive: false,
            },
            flash_config_args: FlashConfigArgs {
                flash_freq: None,
                flash_mode: None,
                flash_size: None,
            },
            flash_args: FlashArgs {
                monitor: false,
                monitor_args: MonitorConfigArgs {
                    monitor_baud: 115200,
                    elf: None,
                    no_reset: false,
                    log_format: None,
                    output_format: None,
                    processors: None,
                },
                ram: false,
                no_verify: false,
                no_skip: false,
                image: ImageArgs {
                    min_chip_rev: 0,
                    mmu_page_size: None,
                    check_app_descriptor: true,
                },
                erase_parts: None,
                erase_data_parts: None,
            },
            image: "reather",
            format: EspIdf,
            idf_format_args: IdfFormatArgs {
                bootloader: None,
                partition_table: None,
                partition_table_offset: None,
                target_app_partition: None,
            },
        },
    ), false
[2025-08-13T04:05:27Z DEBUG] connecting to crates.io:443 at 18.64.67.105:443
[2025-08-13T04:05:27Z DEBUG] No cached session for DnsName("crates.io")
[2025-08-13T04:05:27Z DEBUG] Not resuming any session
[2025-08-13T04:05:27Z DEBUG] Using ciphersuite TLS13_AES_128_GCM_SHA256
[2025-08-13T04:05:27Z DEBUG] Not resuming
[2025-08-13T04:05:27Z DEBUG] TLS1.3 encrypted extensions: [ServerNameAck]
[2025-08-13T04:05:27Z DEBUG] ALPN protocol is None
[2025-08-13T04:05:27Z DEBUG] created stream: Stream(RustlsStream)
[2025-08-13T04:05:27Z DEBUG] sending request GET https://crates.io/api/v1/crates/espflash/versions
[2025-08-13T04:05:27Z DEBUG] writing prelude: GET /api/v1/crates/espflash/versions HTTP/1.1
    Host: crates.io
    User-Agent: ureq/2.12.1
    Accept: */*
    accept-encoding: gzip
[2025-08-13T04:05:28Z DEBUG] Chunked body in response
[2025-08-13T04:05:28Z DEBUG] response 200 to GET https://crates.io/api/v1/crates/espflash/versions
[2025-08-13T04:05:28Z DEBUG] dropping stream: Stream(RustlsStream)
[2025-08-13T04:05:28Z DEBUG] Config: ProjectConfig {
[... 50 lines cut ...]
[2025-08-13T04:05:28Z DEBUG] Re-detected chip: Esp32
[2025-08-13T04:05:28Z DEBUG] Attempting flash enable with: SpiAttachParams { clk: 0, q: 0, d: 0, hd: 0, cs: 0 }
[2025-08-13T04:05:28Z DEBUG] Writing command: SpiAttachStub { spi_params: SpiAttachParams { clk: 00, q: 00, d: 00, hd: 00, cs: 00 } }
[2025-08-13T04:05:28Z DEBUG] Writing command: ReadReg { address: 3ff4201c }
[2025-08-13T04:05:28Z DEBUG] Writing command: ReadReg { address: 3ff42024 }
[2025-08-13T04:05:28Z DEBUG] Writing command: WriteReg { address: 3ff4201c, value: 90000000, mask: None }
[2025-08-13T04:05:28Z DEBUG] Writing command: WriteReg { address: 3ff42024, value: 7000009f, mask: None }
[2025-08-13T04:05:28Z DEBUG] Writing command: WriteReg { address: 3ff4202c, value: 17, mask: None }
[2025-08-13T04:05:28Z DEBUG] Writing command: WriteReg { address: 3ff42080, value: 00, mask: None }
[2025-08-13T04:05:28Z DEBUG] Writing command: WriteReg { address: 3ff42000, value: 40000, mask: None }
[2025-08-13T04:05:28Z DEBUG] Writing command: ReadReg { address: 3ff4201c }
[2025-08-13T04:05:28Z DEBUG] Writing command: ReadReg { address: 3ff42080 }
[2025-08-13T04:05:28Z DEBUG] Writing command: WriteReg { address: 3ff4201c, value: 80000040, mask: None }
[2025-08-13T04:05:28Z DEBUG] Writing command: WriteReg { address: 3ff42024, value: 70000000, mask: None }
[2025-08-13T04:05:28Z DEBUG] Flash detect OK!
[2025-08-13T04:05:28Z DEBUG] Writing command: SpiSetParams { spi_params: SpiSetParams { fl_id: 00, total_size: 400000, block_size: 10000, sector_size: 1000, page_size: 100, status_mask: ffff } }
[2025-08-13T04:05:28Z DEBUG] Writing command: ReadReg { address: 3ff6607c }
[2025-08-13T04:05:28Z DEBUG] Writing command: ReadReg { address: 3ff5a00c }
[2025-08-13T04:05:28Z DEBUG] Writing command: ReadReg { address: 3ff5a014 }
[2025-08-13T04:05:28Z DEBUG] Writing command: ReadReg { address: 3ff5a014 }
[2025-08-13T04:05:28Z DEBUG] Writing command: ReadReg { address: 3ff40014 }
Error:   × Main thread panicked.
  ├─▶ at ~/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/serialport-4.7.2/src/posix/tty.rs:499:9
  ╰─▶ assertion failed: termios2.c_ospeed == termios2.c_ispeed
`

SergioGasquez, 2025-08-13:
I was just able to reproduce the error, Ill do some investigation, in the meantime, using `--baud 9600` seems to be a workaround for the issue

lenzfilipski, 2025-08-24:
I have the same error with espflash, also on arch linux.
Using @sirhcel 's fix did resolve the issue. 
However, I had to migrate `esp-hal` from `0.23.1`  to `1.0.0-rc1` on my project.

sirhcel, 2025-08-25:
> I have the same error with espflash, also on arch linux. Using [@sirhcel](https://github.com/sirhcel) 's fix did resolve the issue.

Thanks for the feedback! I have to state that the fix was authored by @33671 who did all the research and work. ❤️

> However, I had to migrate `esp-hal` from `0.23.1` to `1.0.0-rc1` on my project.

I don't see how this is related to the issue with baud rate setting or reading it back in the first place. Might this be an issue from building the binary you are trying to flash with espflash?

Doloro1978, 2025-08-25:
yay

lenzfilipski, 2025-08-25:
> I don't see how this is related to the issue with baud rate setting or reading it back in the first place. Might this be an issue from building the binary you are trying to flash with espflash?

I just listed what I had to do for completeness. From what I understand it's more related to the difference in version between `esp-hal` and `espflash`.

SergioGasquez, 2025-08-28:
Sorry for the late reply, I was on vacations. Thanks for the fix  @33671 @sirhcel! Just reinstalled `espflash`, which uses the latest `serialport` and the issue is gone!

orhun, 2025-09-01:
Hey all, I was at a conference so I couldn't respond earlier. Thanks for the fix! 

Can we also get a new release of `espflash` with the upgraded dependency?

> Would a patch release of serialport-rs allow you to progress resolving this issue on Arch 

I think now I can patch the dependency if a new release of `espflash` is not on the horizon :)

SergioGasquez, 2025-09-02:
> Can we also get a new release of espflash with the upgraded dependency?

We are working on [`espflash@4.1`](https://github.com/esp-rs/espflash/milestone/4) but we still dont have an estimated date. That being said, since we dont have the `serialport` dependency pinned and it was a patch update, if you uninstall and install `espflash`, it will use the latests version of serial port which contains the fix!